APS-USAXS/livedata #6: waveform string displayed incorrectly

prjemian, 2019-03-20:
Recent change in the [`9idcLAX:USAXS:sampleTitle`]() PV to be a longer string (using a waveform of CHAR) result in a livedata page showing the sample title as `"[0 0 0 0 ..."`

Should show as `""` instead.

prjemian, 2019-03-20:
NOTE: For any PV that is (now) a waveform of CHAR (in the .db file, the record type is `waveform` and this field value `FTVL="CHAR"`), mark in the XML file with additional attribute: ` waveform_char="true"` (the default value is `false`).

jilavsky, 2019-03-20:
Worked for some time, now failed again. What changed? As of 12:37 pm shows zeroes.

prjemian, 2019-03-20:
Just set it from command line with `caput -S 9idcLAX:USAXS:sampleTitle "multi word test"` and this displayed properly for a short time (minutes), then changed to `[109 117 108 116 105  32 119 111 114 100  32 116 101 115 116   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0    0   0   0   0]`.

closed 2019-03-20 by prjemian

prjemian, 2019-03-21:
Observed [live data web page](https://usaxslive.xray.aps.anl.gov/) this morning, verified *this* problem is solved.

commit e3061001 (2019-03-20): fixes #6
commit 8696d604 (2019-03-20): fixes #6
commit 62da7a3a (2019-03-20): fixes #6