APS-USAXS/livedata #24: usaxs.mac file on web page is obsolete

jilavsky, 2019-06-20:
This file: https://usaxslive.xray.aps.anl.gov/usaxs.mac has not changed since April sometimes. This is when using spec code. Permissions due to change to nfs copy?

prjemian, 2019-06-20:
last updated May 23 10:39 (AM)
Permissions look writable
Reason not update is not due to permissions.

Needs further investigation

```
usaxs@usaxscontrol ~/.../USAXS/livedata $ ls -lart /net/joule/export/joule/WEBUSAXS/www_live/ | tail -15
-rw-rw-r--  1 webusaxs usaxs    2119 Apr 20 23:28 Gadikota_Linkam3end.mac
-rw-rw-r--  1 webusaxs usaxs    1916 Apr 22 00:14 Gadikota_NMR_flow.mac
-rw-rw-r--  1 webusaxs usaxs    1915 Apr 22 21:49 Gadikota_Capillary_flow.mac
-rw-rw-r--  1 webusaxs usaxs    5428 May 23 10:39 usaxs.mac
-rwxrwxr-x  1 webusaxs usaxs    7201 May 28 10:26 wwwServerTransfers.py
drwxr-xr-x 26 webusaxs usaxs    4096 Jun 18 13:49 ..
-rw-rw-r--  1 nobody   usaxs    2162 Jun 20 11:25 specmacro.txt
-rw-rw-r--  1 webusaxs usaxs  353294 Jun 20 11:32 scanlog.xml
-rw-r--r--  1 webusaxs usaxs 3735466 Jun 20 11:32 scanlog.log
drwxrwsr-x  5 webusaxs usaxs   12288 Jun 20 11:32 .
-rw-rw-r--  1 webusaxs usaxs   87899 Jun 20 11:32 livedata.png
-rw-rw-r--  1 nobody   usaxs   43451 Jun 20 11:33 report.xml
-rw-rw-r--  1 nobody   usaxs   14709 Jun 20 11:33 index.html
-rw-rw-r--  1 nobody   usaxs   32146 Jun 20 11:33 raw-report.html
-rw-rw-r--  1 nobody   usaxs    5971 Jun 20 11:33 usaxstv.html
```

prjemian, 2019-06-20:
parent directory also group writable

```
usaxs@usaxscontrol ~/.../USAXS/livedata $ ls -lart /net/joule/export/joule/WEBUSAXS/ | tail -15
drwx------   3 webusaxs usaxs      4096 Feb  1  2018 .cache
drwxr-xr-x   2 webusaxs usaxs      4096 Feb  1  2018 Desktop
drwx------   3 webusaxs usaxs      4096 Feb  1  2018 .dbus
drwxr-xr-x   2 webusaxs usaxs      4096 Feb  1  2018 .fontconfig
drwx------   4 webusaxs usaxs      4096 Feb  1  2018 .config
drwx------   2 webusaxs usaxs      4096 Feb  1  2018 .gconf
drwxr-xr-x   3 webusaxs usaxs      4096 Jun 23  2018 .local
drwxr-s---   2 root     usaxs     24576 Jul 31  2018 logs
drwxrwsr-x  17 webusaxs usaxs      4096 May 28 10:00 www
-rw-r--r--   1 webusaxs usaxs     11138 Jun 18 13:46 .history
-rw-------   1 webusaxs usaxs      2496 Jun 18 13:49 .Xauthority
drwxr-xr-x  26 webusaxs usaxs      4096 Jun 18 13:49 .
drwxr-s---   2 root     usaxs     36864 Jun 20 03:59 logs_live
-rw-r--r--   1 webusaxs usaxs 783611455 Jun 20 11:34 .loglogin
drwxrwsr-x   5 webusaxs usaxs     12288 Jun 20 11:34 www_live
```

prjemian, 2019-07-09:
@jilavsky emailed:

> It is convenient to be able to see it when remotely supporting users, any chance of fixing that? 
> 
> This is from spec, but I suspect it is independent if it is spec or BS.

prjemian, 2019-07-09:
@jilavsky emailed:

> I changed the: `/home/beams/USAXS/Documents/eclipse/USAXS/livedata/livedata.xsl` to point to specmacro.txt and now I can see what command file is running. 
> 
> I changed lines 323-331 of this file, you can check if I did it right [...]
> 
> If you look now on https://usaxslive.xray.aps.anl.gov/ and click on spec macro, it will open readable current text file. I feel useful ;-)

prjemian, 2019-07-09:
Thanks for helping me remember what is done.

The SPEC macro file names can be anything and, yes, we gathered a lot of 
crud when we pushed all the various-named macro files to the web site.

Current code does these steps:
1. Copies name of current SPEC file into `report.xml` as XML variable `spec_macro_file`.   Here's an example:

```
<pv id="spec_macro_file" name="9idcLAX:USAXS:macroFile">
   <name>9idcLAX:USAXS:macroFile</name>
   <id>spec_macro_file</id>
   <description>SPEC macro file</description>
   <timestamp>2019-07-09 08:46:28.497603</timestamp>
   <counter>68</counter><units/>
   <value>usaxs.mac</value>
   <raw_value>usaxs.mac</raw_value>
   <format>%s</format>
</pv>
```

2. Copies *contents* of current SPEC file into `specmacro.txt`

3. The XSLT writes the HTML page with the name as provided by the user BUT references the content using, internally, the file name `specmacro.txt`.    While underhanded, this works.

4. all those files, with others, are copied to the www server.

So, the change you made eliminates step 3 above.  If that is clearer to you, then keep the changes.  However, you might consider showing the original name of the macro file on the livedata page.  Somewhere.

prjemian, 2019-07-10:
Let's close this since the revision is acceptable to you.  Open new issue if we see a problem.

closed 2019-07-10 by prjemian