APS-USAXS/livedata #44: Provide remote users with feedback on last set of tunes

jilavsky, 2020-08-10:
Remote users are completely blind to tuning. We are unable to provide them with any presentation of successful tunes. We should develop some kind. of feedback so they at least know what failed and something failed. 
Options: 
1. web  page with plots?
2. PV for each tune with some report back to user? May be "OK" or written statement? 
3. Something smarter enabled by BS?

prjemian, 2020-08-10:
When [our tunes](https://github.com/APS-USAXS/ipython-usaxs/blob/086f5323ccc10ed5fab0cfa11b2fa69bccf6ecd5/profile_bluesky/startup/instrument/plans/axis_tuning.py#L51-L66) run in bluesky, the code assesses whether or not a peak was found:

```
    yield from axis.tune(md=md)
    yield from bps.mv(
        ti_filter_shutter, "close",
        scaler0.count_mode, "AutoCount",
    )

    found = axis.tuner.peak_detected()
    logger.info(f"axis: {axis.name}")
    logger.info(f"starting position: {axis_start}")
    logger.info(f"peak detected: {found}")
    if found:
        logger.info(f"  max: {axis.tuner.peaks.max}")
        logger.info(f"  center: {axis.tuner.peaks.cen}")
        logger.info(f"  centroid: {axis.tuner.peaks.com}")
        logger.info(f"  fwhm: {axis.tuner.peaks.fwhm}")
    logger.info(f"final position: {axis.position}")
```

This might be a place where we can add source such diagnostics.  The name of the motor being tuned is provided in `axis.name`.

prjemian, 2020-08-10:
So, easiest would be a `bo` PV for each axis tuned that says peak found True or False.  

A plot could be generated from the bluesky (tuning) run and could be placed in an axis-dependent plot (such as `tune_mr.svg`) on the livedata page.