APS-USAXS/livedata #45: raw data directory changes in 2020-09

prjemian, 2020-09-09:
In https://github.com/APS-USAXS/ipython-usaxs/pull/379, the raw data directory will change according to the table described here: https://github.com/APS-USAXS/ipython-usaxs/issues/361#issuecomment-688597657

update:

* [ ] livedata
* [ ] specplots

prjemian, 2020-09-16:
FlyScan is not plotted when collected by SPEC.  This is because the number of SPEC comments has changed and the `livedata` was expecting the HDF5 file name to be provided at a fixed number of comments from the scan start.

```
        comment = scan.comments[2]
        key_string = 'FlyScan file name = '
```

but the data file now has

```
#C splB_8
#C Wed Sep 16 16:37:58 2020.  Need to fix FIlename Format when other ADs are running.
#C Wed Sep 16 16:37:58 2020.  Collected FLY2 jpg data.
#C Wed Sep 16 16:37:58 2020.  FlyScan #260 started.
#C Wed Sep 16 16:37:58 2020.  FlyScan file name = ./09_16_Krzysko/09_16_Krzysko_usaxs/splB_8_0059.h5.
#C Wed Sep 16 16:37:58 2020.  Please wait approximately 135s to complete FlyScan..
#C Wed Sep 16 16:40:09 2020.  Flyscan done after 129s..
#C Wed Sep 16 16:40:09 2020.  FlyScan finished.
#C Wed Sep 16 16:40:09 2020.  Moving AR motor back to 8.74763 degrees (beam center).
#C Wed Sep 16 16:40:09 2020.  splB_8.
#C Wed Sep 16 16:40:28 2020.  Ready for SAXS mode.
#C Wed Sep 16 16:40:28 2020.  Need to fix FIlename Format when other ADs are running.
#C Wed Sep 16 16:40:28 2020.  Collected FLY2 jpg data.
#C Wed Sep 16 16:40:28 2020.  *************   Started SAXS Exposure   **********************.
#C Wed Sep 16 16:40:28 2020.  First Sample Name ./09_16_Krzysko/09_16_Krzysko_saxs/splB_8_0059.hdf.
```

The `livedata` code needs to search the comments for a match rather than assume the match is one specific comment.

prjemian, 2020-09-17:
The `specplots` page is not getting the new directories when SPEC runs.

prjemian, 2020-09-17:
`specplot` was failing on a reported SPEC file name that was actually the directory that contains a SPEC data file.  For now, let's just log that the reported file name is actually a directory, then proceed to the next file name in the _current_ list.

prjemian, 2020-09-17:
Directories are now created but the FlyScan data files are not found for display.  Upstream fix for the SPEC code will be added later today and it is expected all data will then be visible.

prjemian, 2020-09-17:
Same for FlyScan plots on main `livedata` page.  Not shown until upstream fix for SPEC code is added.

jilavsky, 2020-09-18:
IT'S working!

prjemian, 2020-09-18:
Right, FlyScan plot is now shown on livedata page.  NOT shown on specplots page.  File path to `.h5` file looks right, in subdirectory from SPEC data file.  Need to check the specplot code.