APS-USAXS/livedata #49: Improve Plot look

jilavsky, 2020-09-25:
Remote users requested recently modifications to plot of last 10 data sets. Something to consider, if it can be done easily.
1. Move legend top right corner and make smaller font
2. Remove S numbers (Users do not understand them anymore)

The good news is, that this clearly shows users need this image and are using it.

prjemian, 2020-09-25:
I believe the legend location under our control, as is the `S` character.  Before using absolute positioning, I might see if MatPlotLib can use its own smarts to position the legend.

The legend was placed in lower-left corner since the box width is variable (depends on the longest length label text) so it is harder to position the top-right corner of the box.  Lower-Left is deterministic.

prjemian, 2020-09-25:
The long length of the current labels is exceptional.
[image]

Tell the users to write their abstracts somewhere else.

prjemian, 2020-12-14:
It may be possible to ask MatPlotLib to auto-place the legend box.  Perhaps using the [`tight_layout`](https://matplotlib.org/3.1.3/tutorials/intermediate/tight_layout_guide.html) method or perhaps not specifying the `loc=` kwarg: https://github.com/APS-USAXS/livedata/blob/089e75cd97af02d549904f99019d7e0bb072c0b8/plot_mpl.py#L85

prjemian, 2021-10-08:
This was easier than expected.  MatPlotLib will decide where to place the legend (in the most open space).

commit 29665300 (2021-10-08): FIX #49
closed 2021-10-08 by prjemian