APS-USAXS/livedata #52: Fix web page and live data records for waveform

jilavsky, 2021-08-02:
9idcLAX:specFile can get too long for string in epics. We need to change to ~90 characters waveform. Unluckily, when we tried this it breaks web page and live data page. Apparently Python interprets this not as string but array of ASCII codes. 
We need to fix this so staff and users do not have to follow number of used characters too much. 
I will change the PV August 3 and this can be tested and fixed in August shot down.

jilavsky, 2021-09-13:
@prjemian this was not addressed. Is this going to be issue for BlueSky?

prjemian, 2021-09-13:
https://github.com/APS-USAXS/scanLog

prjemian, 2021-09-13:
https://github.com/APS-USAXS/livedata

prjemian, 2021-09-13:
In those two other projects, fixes were applied to get the PV value as a string waveform instead of showing it as a byte string.

closed 2021-09-13 by prjemian