APS-USAXS/livedata #54: Live data page not working with step scanned BS data

jilavsky, 2021-09-14:
Running step scanning in BS (12_03_Weber), collecting hdf5 files. LiveData page is not working with them. 
Looking at web page for scanlog, records seem to be OK, so this is likely graph generator not working? Probably was not updated to handle step scanning data?

prjemian, 2021-09-14:
same for `/share1/USAXS_data/2021-09/09_13_Lisa1/09_13_Lisa1_usaxs/GlassBlank_0067.h5`

prjemian, 2021-09-14:
This will be used a lot in the next weeks.  Get it working.

prjemian, 2021-09-17:
Aha!  The code that processes `uascan` info expects data that was written by SPEC.  The first failure is that the SPEC scan object does not have a `metadata` attribute written from the Bluesky SPEC data file writer callback.  Here is the line that fails: https://github.com/APS-USAXS/livedata/blob/84736385714c8a7f73f8cd6aa9219d8542ad3a7f/calc.py#L202

Now that we have a path and HDF5 file name, we can find the data needed for the data reduction (angle, counts -> I v Q) routine: https://github.com/APS-USAXS/livedata/blob/84736385714c8a7f73f8cd6aa9219d8542ad3a7f/calc.py#L195

It will take some work to extract that from the raw scan data.

prjemian, 2021-09-17:
Look for the lines that start with `# MD hdf5_`

prjemian, 2021-09-17:
This is a start to find the HDF5 file:

```python
In [31]: sdf = SpecDataFile("/share1/USAXS_data/2021-09/09_14_Juliane/09_14_Juliane.dat")

In [32]: scan = sdf.getScan(226)

In [33]: scan.interpret()

In [34]: [line for line in scan.raw.split("\n") if line.startswith("#MD hdf5_")]
Out[34]:
['#MD hdf5_file = CMFe2_7_0051.h5',
 '#MD hdf5_path = /share1/USAXS_data/2021-09/09_14_Juliane/09_14_Juliane_usaxs']
```

prjemian, 2021-09-17:
@jilavsky I **added** a test for BS-created SPEC files (test could be improved by looking at the header lines instead of testing of `metadata` attribute is present).  If the `metadata` attribute is not found in the spec scan object, then assume there is an HDF5 data file specified in the `#MD ` and get the uascan data from the HDF5 file (if found).

It seems (to me) easier to test at run time, if you can do this, otherwise, I'll engineer an offline test with existing data.

prjemian, 2021-09-17:
Here's the scan I used for testing:

```python
In [135]: sdf.fileName
Out[135]: '/share1/USAXS_data/2021-09/09_14_Juliane/09_14_Juliane.dat'

In [136]: sds.scanNum
Out[136]: '226'
```

commit 834226bd (2021-09-18): FIX #54 re-arrange
closed 2021-09-18 by prjemian