APS-USAXS/livedata #57: Add (or fix) live data BS step scans

jilavsky, 2022-02-20:
Now, when BluSky step scanning works fine, we should add live data display for these data. It is Nexus file with some differences. Should not be difficult to modify existing code for Flyscanning. 
Data Collected 2-20-22 around 4:30-6pm, file names with "step" in them.

prjemian, 2023-01-13:
One reason this issue has been hard to diagnose the root cause is that the VS Code debugger stopped supporting Python2.7.  At least for a time, there is a [work around](https://stackoverflow.com/questions/72214043/how-to-debug-python-2-7-code-with-vs-code) but this is not expected to remain a possibility long term.

Very important to continue on #48!

prjemian, 2023-01-13:
To help diagnose this problem offline, need a conda/[micromamba](https://mamba.readthedocs.io/en/latest/user_guide/micromamba.html) environment with the correct package versions.  This one has worked so far on a couple example runs from 2022-12-17 & 18 to determine that `calc.reduce_uascan(sds)` does not raise any exceptions.

```yml
name: py2usaxs

# use:
#   date; micromamba create -y -n py2usaxs -f ./py2usaxs-env.yml; date
#   # then activate with:
#   eval "$(micromamba shell hook --shell=bash)"
#   micromamba activate py2usaxs

channels:
  - conda-forge
  - defaults
  # - aps-anl-tag
  # - apsu
  # - sveseli

dependencies:

  # --- Python core packages
  - python =2
  - h5py
  - lmfit
  - lxml
  - matplotlib
  - paramiko
  - pyepics
  - scipy
  - spec2nexus

  - ipython
  - jupyter

  # - pip:
  #   - happi
```

prjemian, 2023-01-13:
Next is to examine in `scanplots.py`:

```bash
(py2usaxs) prjemian@zap:~/.../USAXS/livedata$ git grep reduce_uascan
calc.py:    return reduce_uascan(sds)
calc.py:def reduce_uascan(sds):
scanplots.py:    usaxs = calc.reduce_uascan(scan.spec_scan)
```

prjemian, 2023-01-13:
One problem with this line: https://github.com/APS-USAXS/livedata/blob/f92ccd5081b2b492401c056eebc9a752c23acabe/scanplots.py#L390

```py
AttributeError: 'SpecDataFileScan' object has no attribute 'spec_scan'
```

prjemian, 2023-01-13:
This line should change `scan.spec_scan` to just `scan`.  What about elsewhere?

```bash
(py2usaxs) prjemian@zap:~/.../USAXS/livedata$ git grep "scan.spec_scan"
scanplots.py:            if scan.spec_scan is None:
scanplots.py:            if scan.spec_scan is None:
scanplots.py:            if not scan.spec_scan.__interpreted__:
scanplots.py:                    scan.spec_scan.interpret()
scanplots.py:            hdf5_file = get_Hdf5_Data_file_Name(scan.spec_scan)
scanplots.py:        title = scan.spec_scan.comments[0]
scanplots.py:    usaxs = calc.reduce_uascan(scan.spec_scan)
```

prjemian, 2023-01-13:
Seems to be some confusion in use of `scan` and `scan.spec_scan` and also `scan.scanNum` vs. `scan.scan_number` in `scanplots.get_USAXS_uascan_ScanData()`.  Additional use in other parts of `scanplots.py`.  Will only change within the one function for now.

Keep this in mind for #48.

prjemian, 2023-01-13:
Here is the root cause of the confusion noted above: https://github.com/APS-USAXS/livedata/blob/f92ccd5081b2b492401c056eebc9a752c23acabe/scanplots.py#L103-L115

prjemian, 2023-01-13:
`AttributeError: Unknown property axisbg`: https://github.com/APS-USAXS/livedata/blob/f92ccd5081b2b492401c056eebc9a752c23acabe/plot_mpl.py#L52

```bash
(py2usaxs) prjemian@zap:~/.../USAXS/livedata$ micromamba list matplot
List of packages in environment: "/home/prjemian/micromamba/envs/py2usaxs"

  Name             Version  Build           Channel    
─────────────────────────────────────────────────────────
  matplotlib       2.2.5    ha770c72_3      conda-forge
  matplotlib-base  2.2.5    py27h250f245_1  conda-forge
```

prjemian, 2023-01-13:
solution: https://stackoverflow.com/questions/50504053/attributeerror-unknown-property-axisbg

PR #60 (2023-01-15): plot uascan again on livedata page
- close #57
closed 2023-01-15 by jilavsky